gio-dot/Z1-Mini #1: Estimated battery life

A7F, 2021-07-09:
Hi!
I just jumped into the iot world and I discovered this small module because I was searching an alternative to the esp01, as it is a bit too power hungry for my use case. Do you by any chance know what should be an estimated battery life of this z1 mini if we plug only one sensor, let's say the bmp280 and read like every 10mins? Asking because the aqara sensors last like 1.5 years on a single cr2032 cell, so I was wondering if it was something doable or not.

Gio-dot, 2021-07-09:
> Hi!
> I just jumped into the iot world and I discovered this small module because I was searching an alternative to the esp01, as it is a bit too power hungry for my use case. Do you by any chance [...]

Hi, i know that this question may seem against my interests but: why won't you use aqara/mija sensors for this simple battery powered task?
Z1-Mini is more indicated for powered/mixed/complex devices. 
In any case, for battery powered device you have to use ptvo paid firmware version. Battery life is similar to Aqara/Mija sensors.

A7F, 2021-07-10:
to be honest, it's mostly because I would use it to build a multisensor, but I simply needed a rough estimate if I can expect a similar battery life compared to the Aqara. I never had one and I never had the chance to actually see what they have inside, but my guess is that there is a cc2530 and a single sensor, maybe 2 (for example their motion sensor also has a LDR inside if I remember correct?) and still they get a  1.5yr battery life. This one is based on the TI-E18 which in turn has a cc2530, so... this is why I asked.
Thank you anyway :)

closed 2023-01-17 by Gio-dot