gio-dot/Z1-Mini #2: measurement interval unclear

wspeth, 2021-09-11:
sorry to bother but from the published material I cannot figure out at which frequency the values are measured and transmitted.

From https://ptvo.info/zigbee-configurable-firmware-features/premium/ it appeares some LED is blinking at either 5 seconds intervals or at 30 seconds intervals but I doubt this is the same for measurement.

can you shed some light into the behaviour please?

Gio-dot, 2021-09-12:
Hi, here is the pkace where you can set reporting interval:
[image]

wspeth, 2021-09-12:
oh, got it
so it is one second to 65536 seconds (>18h). Great.
Thanks a lot
will go to Italy in a week (Giglio) and have an espresso celebrating your
work ;-)

Walter

Gio-dot, 2021-09-12:
Cheers. You are welcome ;)

closed 2021-10-15 by Gio-dot