gio-dot/Z1-Mini #6: Zigbee pairing

patrickjane, 2022-05-31:
Hello,
I have ordered a Z1-Mini, and I am currently reading through the documentation. I undestand how to generate the firmware and how to flash it. However, how does the zigbee pairing process work?

branitelj91, 2023-01-11:
Gio. 

I am, also, interested on how to put Z1 mini into pairing mode. 
Please answer

Gio-dot, 2023-01-17:
To force a new pairing there are 3 ways:
Option 1
CC253x: Re-flash firmware
Option 2
CC2530: Power on, wait 2 seconds, power off, repeat this cycle three times.
CC2531: Power on, press and hold down the S2 button for 5 seconds.
Option 3
CC2530: Power on, press and hold down the button on “Input 1” for 10 seconds.

branitelj91, 2023-01-20:
Thank You. 

I managed to pair it and make it work without Zigbee2Mqtt, (was working directly through CONBEE coordinator).